bigtreetech/Manta-E3EZ #1: How to setup the internal CAN interface?

rclarke250, 2023-01-16:
config says CAN bus (on PB12/PB13) and that is not a choice in make menuconfig.  So how does one setup the internal CAN interface with a CB1 mounted on the board?

emersonsc, 2023-01-30:
Looks like there is a pull request but Kevin being Kevin has an issue with something stupid and its being held up.

looxonline, 2023-01-30:
For now, you'll need to use the BTT klipper branch where it is supported until Kev upstreams it. https://github.com/Klipper3d/klipper/pull/5981

JamesH1978, 2023-02-01:
The issue is that the files submitted in the PR https://github.com/Klipper3d/klipper/pull/5981 , contain sections for [tmc2240] module which does not exist in mainline klipper. Until btt submits this modile to add the functionality the PR will probably stay stale.

Thanks
James

mar-cun, 2023-02-03:
Any update/idea on when this might get incorporated? I am trying to upgrade my printer right now with the E3EZ and an EBB42 (not that I can find any specific instructions on how to accomplish that, so I have an email into tech support). So in the interim I ended up copying the two updated files from the pull request (kconfig & fdcan.c) and overwriting them in my klipper folders and I was successful in getting the PB12/PB13 options to show up, but not sure if something else is needed. When trying to query for the can bus ID at the end of the setup I get a device not found. But not sure if that's just something else I'm doing wrong, or if it's a problem with the environment.

rclarke250, 2023-02-08:
The latest Klipper3d update allowed me to get CAN functioning.  A setup guide in the manual would be helpful for those not familiar with using USB to CAN bridge mode.

closed 2023-02-09 by rclarke250

hauserkristof, 2023-08-04:
Hi @rclarke250 !

Can you help me with enabling the CAN Bus on the device? I'm struggling with it a bit.

Thank you in advance,
Kristóf

rclarke250, 2023-08-05:
> Hi @rclarke250 !
> 
> Can you help me with enabling the CAN Bus on the device? I'm struggling with it a bit.
> 
> Thank you in advance, Kristóf

Sure what do you need help with?

yourbuddydinec, 2023-08-08:
> > Hi @rclarke250 !
> > Can you help me with enabling the CAN Bus on the device? I'm struggling with it a bit.
> > Thank you in advance, Kristóf
> 
> Sure what do you need help with?

I was curious what you chose for the pins on katapult/Canboot. It looks like on the E3EZ it should be the PC12/PB13 but the only choice in Canboot is PD12/PD13

rclarke250, 2023-08-08:
I didn't use Canboot. I just used Klipper make menuconfig...set the STM device to USB bridge mode, and set the pins for can, make and copied to a microsd after changing klipper.bin to firmware.bin and booting the machine.

Deapper, 2023-08-13:
> I didn't use Canboot. I just used Klipper make menuconfig...set the STM device to USB bridge mode, and set the pins for can, make and copied to a microsd after changing klipper.bin to firmware.bin and booting the machine.

Do you use an EBB tool head? I setup the MCU for canbus but could not see the ebb42 I have.

I followed this guide https://www.youtube.com/watch?v=EA-oBfenxAE&t=160s

rclarke250, 2023-08-13:
Yes, I actually have 2 EZ boards, one on an Ender 3 pro with CAN to an
EBB42, and another in an Artist-D pro running 2 EBB42 in Idex with BTT
H2's.

Deapper, 2023-08-13:
So do you use the following under make menuconfig?
USB bridge - CANBUS - PB12 PB13?
Both Manta and EBB42 usb to (CAN bus (on PB0/PB1)?

Also what baudrate do you use? 1000000?

And do you have the 120omh pin shorted on both Manta and EBB42?

thanks

rclarke250, 2023-08-14:
The Manta yes, USB-Canbus-PB12,PB13.
The EBB Canbus PB0,PB1
I used Baud rate 500000. Doesn't seem to be an issue.
And yes, 120 Ohm jumper on each.
cable has to be twisted, with High to High, Low to Low. Doesn't need to have a lot of twist,  I also twisted my Power cables with it, and is working okay.