bigtreetech/Manta-E3EZ #2: How To put board in DFU Mode

emersonsc, 2023-01-30:
the usual reset + boot button method does NOT put the board in DFU mode, making it impossible to update firmware via STM32Programmer

DarKingF0x, 2023-01-30:
Right next to the front USB there is a dip switch that you need to switch to the opposite direction in order to use the DFU mode it's in the manual

looxonline, 2023-01-30:
> Right next to the front USB there is a dip switch that you need to switch to the opposite direction in order to use the DFU mode it's in the manual

The switch next to the micro USB connector is related to the interface that is presented to the micro USB port by the CB1 and will not impact DFU mode on the MCU. One direction offers a direct USB connection which allows downloading firmware onto an embedded eMMC and the other allows a UART connection via a converter IC which allows serial debugging or serial terminal commands.

DarKingF0x, 2023-01-30:
Ssh into board via serial connection, run kiuh, and flash

dogjutsu, 2023-03-05:
I was able to un-brick my Manta E3 EZ firmware from the OS on the CB1 -- this is what I did, not sure if everything being unplugged is necessary:

- Board with CB1 installed (OS already installed on installed SD Card) completely unplugged from everything, USB power jumper applied.
- Hold down BOOT button (middle of the board near the middle stepper driver) while plugging in USB power, release BOOT button once powered.
- SSH into CB1, confirm dfu mode vis 'dmesg | grep -A4 dfu' -- should see something like:

```
[    3.591220] usb 2-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    3.598544] usb 2-1.1: Product: DFU in FS Mode
[    3.598552] usb 2-1.1: Manufacturer: STMicroelectronics
[    3.603853] Freeing unused kernel memory: 2176K
[    3.608236] usb 2-1.1: SerialNumber: <REDACTED>
```

- Install 'fwupd' package:  'sudo apt-get update ; sudo apt-get install fwupd'
- Optionally, further confirm DFU mode via 'fwupdmgr get-devices':
```
biqu@klipper:~/klipper$ fwupdmgr get-devices
WARNING: UEFI capsule updates not available or enabled in firmware setup
  See https://github.com/fwupd/fwupd/wiki/PluginFlag:capsules-unsupported for more information.
BQ-H616
│
└─DFU in FS Mode:
      Device ID:          <REDACTED>
      Current version:    2.0
      Vendor:             STMicroelectronics (USB:0x0483)
      GUIDs:              <REDACTED>
                          <REDACTED>
      Device Flags:       • Updatable
                          • Is in bootloader mode
```

- Flash klipper.bin with dfu-util (serial from dmesg above):
```
biqu@klipper:~/klipper$ sudo dfu-util -D ./out/klipper.bin --serial <REDACTED> -a 0 -s 0x8002000:leave -R
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 0483:df11
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 1024
DfuSe interface name: "Internal Flash   "
Downloading to address = 0x08002000, size = 26664
Download        [=========================] 100%        26664 bytes
Download done.
File downloaded successfully
dfu-util: Error during download get_status
```

- After a few seconds, your `/dev/serial/by-id/usb-Klipper_stm32g0b1xx_XXXXXXX-if00` device should re-appear.  You can also grep for 'Klipper' in dmesg to confirm:  `dmesg | grep -C5 -i klipper`

Hopefully this helps unbrick somebody else unbrick their firmware without resorting to ST-Link (which I couldn't get to work, fwiw).

dogjutsu, 2023-03-05:
Addendum: The above didn't 'stick' after reboot, however I was able, with everything plugged in, to enter DFU mode by holding down the BOOT button while powering on, then this worked and stuck:

make flash FLASH_DEVICE=0483:df11

dogjutsu, 2023-03-05:
Update:  My mistake, that apparently does not survive power cycle either.

looxonline, 2023-03-05:
> Update: My mistake, that apparently does not survive power cycle either.

What do you mean that it does not survive a power cycle? Does the MCU no longer appear to the CB1 over the USB link after a power cycle?

dogjutsu, 2023-03-05:
> > Update: My mistake, that apparently does not survive power cycle either.
> 
> What do you mean that it does not survive a power cycle? Does the MCU no longer appear to the CB1 over the USB link after a power cycle?

So immediately post `make flash` after a few seconds the klipper device appears under `/dev/serial/by-id/usb-Klipper_stm32g0b1` and I can start the Klipper service and everything works -- but after rebooting the /dev device doesn't show up and no sign of the device in lsusb / dmesg (unless I boot to DFU mode).

I suspect what's going on is although I flashed the actual Klipper firmware, the bootloader is hosed, so there is nothing to load correctly boot the Klipper firmware.

I'm going to see if there's a way to initiate whatever reset the dfu-util is doing without actually flashing, then I can just make that happen on boot.

looxonline, 2023-03-05:
> So immediately post `make flash` after a few seconds the klipper device appears under `/dev/serial/by-id/usb-Klipper_stm32g0b1` and I can start the Klipper service and everything works -- but after [...]

Why don't you just reinstall the bootloader? Install canboot and set it to operate in USB mode.

dogjutsu, 2023-03-05:
> Why don't you just reinstall the bootloader? Install canboot and set it to operate in USB mode.

I've been poking around for a bootloader bin/hex/etc that'll work for the stm32g0b1 ... happen to know where to source one?

looxonline, 2023-03-05:
> I've been poking around for a bootloader bin/hex/etc that'll work for the stm32g0b1 ... happen to know where to source one?

Canboot is the way to go https://github.com/Arksine/CanBoot

dogjutsu, 2023-03-05:
> Canboot is the way to go https://github.com/Arksine/CanBoot

Thanks, I'll see what I CAN do!

dogjutsu, 2023-03-05:
> > Canboot is the way to go https://github.com/Arksine/CanBoot
> 
> Thanks, I'll see what I CAN do!

@looxonline thanks a bunch for the tip, I'm good to go now!  

For those playing along at home, basically followed the instructions here:
https://github.com/Arksine/CanBoot/issues/61#issuecomment-1405721325

...and then remembered to setup my printer.cfg accordingly:

```
[mcu]
canbus_uuid: <redacted>
```

looxonline, 2023-03-05:
> > > Canboot is the way to go https://github.com/Arksine/CanBoot
> > 
> > 
> > Thanks, I'll see what I CAN do!
> 
> @looxonline thanks a bunch for the tip, I'm good to go now!
> 
> For those playing along at home, basically followed the instructions here: [Arksine/CanBoot#61 (comment)](https://github.com/Arksine/CanBoot/issues/61#issuecomment-1405721325)
> 
> ...and then remembered to setup my printer.cfg accordingly:
> 
> ```
> [mcu]
> canbus_uuid: <redacted>
> ```

Good stuff!

Tina-Builds, 2023-05-08:
@looxonline
Do you know where to find the original bootloader, eg. if someone has installed CanBoot and wants to go back?

killrmodz, 2023-06-25:
> @looxonline thanks a bunch for the tip, I'm good to go now!
>
> For those playing along at home, basically followed the instructions here: [Arksine/CanBoot#61 [...]

Followed all of the instructions above, and my firmware seems to flash but doesn't as a powercycle won't allow klipper to come up without putting back in the microsd.  So i'm lost here and am looking for guidance.  Thanks!

Valiance, 2023-08-23:
> Followed all of the instructions above, and my firmware seems to flash but doesn't as a powercycle won't allow klipper to come up without putting back in the microsd. So i'm lost here and am looking [...]

I have to drive with my SD card inserted in the Manta E3 EZ board.

looxonline, 2023-08-23:
> @looxonline Do you know where to find the original bootloader, eg. if someone has installed CanBoot and wants to go back?

Completely missed this but the bootloader is in the files section.