bigtreetech/Manta-E3EZ #3: Ender 3 v2 Enclosure Damages SD Card

cmcfalls2, 2023-02-06:
I noticed that the mainboard sd card slot directly aligns with the E3v2 mounting pin location. If I reattach my electronics enclosure cover it is going to tear the SD card slot off the board. Seems like a bit of an oversight. Either move the SD card location slightly or push it further underneath the board since it can't be accessed with the board installed anyway. Is there an update/fix coming for this? I don't want to install it in my other v2's if there is a revision coming.

looxonline, 2023-02-14:
See enclosure STLs that have been uploaded for the E3pro/V2 and 5.

blandry80, 2023-02-16:
> See enclosure STLs that have been uploaded for the E3pro/V2 and 5.

Can someone resubmit the file for the front enclosure, all I am getting is the raw code? Thank you in advance.

looxonline, 2023-02-17:
> > See enclosure STLs that have been uploaded for the E3pro/V2 and 5.
> 
> Can someone resubmit the file for the front enclosure, all I am getting is the raw code? Thank you in advance.

You may need to right click on the download link and say "Save file As"

markewert, 2023-04-03:
Even with 'Save file As' the file is corrupt. Since without this the Manta E3EZ cannot be used, I am going to have to return it which is annoying.

looxonline, 2023-04-03:
> Even with 'Save file As' the file is corrupt. Since without this the Manta E3EZ cannot be used, I am going to have to return it which is annoying.

It's really just this simple. I just tried it and it works perfectly. https://www.gitkraken.com/learn/git/github-download#how-to-download-a-file-from-github

[image]

markewert, 2023-04-03:
Partial thanks to @looxonline . I download files off GitHub all the time but there's a trick to this one and Luke's "simple" instructions are only partially accurate. After you click the filename from the list there will not be a 'Raw' button. The 'Download' button next to the file also does not work normally (this is what should be fixed). Instead, right-click on the Download button and choose 'Save Link As'. The file that downloads will not be corrupt.

looxonline, 2023-04-03:
> Partial thanks to @looxonline . I download files off GitHub all the time but there's a trick to this one and Luke's "simple" instructions are only partially accurate. After you click the filename [...]

I grabbed a screenshot of the "Raw" button as I was making that response. It's exactly what was on my screen. Maybe your browser does something different with the file. In any case, in my earlier response I also mentioned the "download link" which would apply to your "download" button.

Dids, 2023-06-24:
The download issues are due to Git LFS. Downloading the zip archive of the entire repository should include all the files, but alternatively you can also install both git and git-lfs and clone the repository.