bigtreetech/Manta-E3EZ #4: Ender 3 Pro support

emersonsc, 2023-02-06:
Will there ever be support for the Ender 3 Pro? The board does not fit in the Stock "upside down" electronics box unless you have no plans on using the cat5 or usb connections.

looxonline, 2023-02-14:
It is exactly the same as the E3V2.

DarKingF0x, 2023-02-27:
Same on regular E3. Will have no access to the HDMI, USB or the network cable. I absolutely love this board just wish I could fit it in there and be able to use it

looxonline, 2023-02-27:
> Same on regular E3. Will have no access to the HDMI, USB or the network cable. I absolutely love this board just wish I could fit it in there and be able to use it

This would likely get you sorted for HDMI on the regular Ender 3: https://www.adafruit.com/product/3557

emersonsc, 2023-02-27:
> Same on regular E3. Will have no access to the HDMI, USB or the network cable. I absolutely love this board just wish I could fit it in there and be able to use it

Not sure about that. Once I used the files Luke pointed out, I have access to the network cable on me 3Pro. I don't use the HDMI so I can't say because I didn't pay attention.

> This would likely get you sorted for HDMI on the regular Ender 3: https://www.adafruit.com/product/3557

That's nifty, thanks!

I'm going to mark this closed as Luke pointed out the files work for the 3 Pro as well

closed 2023-02-27 by emersonsc

docdawning, 2023-04-18:
I'd like BTT to also offer an all metal chassis option to fit this board on an E3 chassis.