bigtreetech/Manta-E3EZ #5: Manta E3EZ - CM4 - Camera/DSI ports not working

Will-wastelander, 2023-02-24:
Unable to get DSI screen or rPi cam to work on this board. I have copied the dt-blob.bin file to /boot as stated in documentation. What other steps are required to get these items working ? Screen is BTT PiTFT50 v2.0, and Camera is rPi NoIR camera. Camera worked on rPi 4 4GB before switching to Manta E3EZ mainboard. I have already replaced the cables with brand new ones, and confirmed that they are connected correctly.

Will-wastelander, 2023-02-26:
Got display working, but camera is still not working.

Will-wastelander, 2023-02-27:
This is my second board that is doing the same thing.

DarKingF0x, 2023-02-27:
Using a CM4 or CB1?

DarKingF0x, 2023-02-27:
Cm4 should use hdmi, for this particular board the CB1 uses the SPI DSi, and CM4 hdmi

Will-wastelander, 2023-02-27:
I have DSI display working w/ CM4, I need to get rPi camera working now.

looxonline, 2023-03-06:
Just copying a pic of your comment on Discord over to here for reference:

[image]

Will-wastelander, 2023-03-06:
Determined that a CSI  "A type" cable is required in order for the camera to work, as the pins are in fact incorrect on the Manta E3EZ. The below 2 items are what I had to use in order for a standard 2 foot CSI cable to work. The pins are mirrored from what the standard CSI pins are.

https://www.amazon.com/uxcell-Flexible-Cable-Ribbon-Player/dp/B07RVD2MZV/
https://www.amazon.com/gp/product/B07RWF4QCY/

karrui, 2023-03-22:
If you do not want to buy a new type A FFC, you could also peel away the plastic backing of one end and fold the now soft connectors upwards to form a pseudo type A FFC from a type B.

looxonline, 2023-03-23:
> Determined that a CSI "A type" cable is required in order for the camera to work, as the pins are in fact incorrect on the Manta E3EZ. The below 2 items are what I had to use in order for a standard [...]

As we discussed on Discord, I have confirmed that the pins on the Manta E3EZ are not inverted and there are simply different connectors used between it and the Pi4b adapter which is why the numbering looks different. I can also prove this based on not my evidence, but the evidence that you have provided. See below.

You state that you fixed this issue by purchasing a type A cable and an extender and using them in combination with a standard, 2 foot CSI cable. Let's put this statement to the test and see if it results in an overall type A or type B cable.

Some important things to note are:
- The extender only has contacts on one side.
- The extender is straight through.

Drawing the cable type using ASCII art below reveals something interesting:

TOP ---------2ft TypeB------------- BOTTOM (EXT IN) -- (EXT OUT) BOTTOM -----------TypeA--------- BOTTOM

We note that the combination you have mentioned actually results in an overall type B and *not* a type A. Since you state that this works, this proves that the Manta E3EZ pins are correct.

I would have to assume that either you were originally using the extender with two type B cables or you were originally using a single type A. Either way, it is not cool to mislead others into thinking that they need to purchase a type A cable for the board to work. If this was done by mistake, then that is understandable because it is easy to get lost in the details of which side of the connector should go where and what cable is what type. Nevertheless, I want other reading this thread to please take note of this comment so that they do not purchase a type A cable only to find it not working.

Additional evidence of the extender flipping the type is found in the amazon reviews for it:

[image]

Will-wastelander, 2023-04-15:
So after 4 boards, I was able to get this working with my original setup.

closed 2023-04-15 by Will-wastelander

Gabbajoe, 2023-10-18:
> If you do not want to buy a new type A FFC, you could also peel away the plastic backing of one end and fold the now soft connectors upwards to form a pseudo type A FFC from a type B.

Hi this hint made for me on a Manta M5P the raspberry Pi cam v3 working! If I use the Type B cable the CM4 module on my Manta M5P gose away. this means my shells closes and ping to the host is no longer possible also the the activity LED was just dim flickering and the CM4 module never boots. So I the cam connector on the Manta M5P is wrong.

jevermeister, 2025-01-10:
The pins are not reversed. It's just that the CSI connector needs to have the cable the other way around. Use a regular cable (Type B / reverse, so the blue ribbon is on a different side on each end) just as the ones which are shipped with e.g. the Raspberry Camera Pi 3 and insert them according to this picture. If you buy a Type A or peel of the plastic of your type B and wrap it around, you will shorten your 3.3V rail and the Pi won't boot up. Luckily it did not damage anything for me.

[image]