bigtreetech/Manta-E3EZ #8: Manta E3EZ with rp3 connection

bmorgan99, 2023-03-13:
I get no communication when trying to connect via raspberry pi3b w/USB cable. 

firmware uploads,  3V3 is green, firmware.bin renamed
tried 3 cables, all USB ports
removed, reinstalled klipper (kiauh) on pi
no unique by-id serial device is created
normal serial devices don't work, i.e. serial1, ttyAMA0

menuconfig:
```
STM32G0B1
8KiB bootloader
8 MHz crystal
USB (on PA11/PA12)
```
systemctl status klipper.service
```
 klipper.service - Klipper 3D Printer Firmware SV1
     Loaded: loaded (/etc/systemd/system/klipper.service; enabled; vendor preset: enabled)
     Active: active (running) since Sat 2023-03-11 06:36:18 CST; 2h 4min ago
       Docs: https://www.klipper3d.org/
   Main PID: 814 (python)
      Tasks: 2 (limit: 1472)
        CPU: 36.829s
     CGroup: /system.slice/klipper.service
             └─814 /home/pi/klippy-env/bin/python /home/pi/klipper/klippy/klippy.py /home/pi/printer_data/config/printer.cfg -I /home/pi/printer_data>
```

KrisWragg, 2023-03-13:
As discussed on biqu community I have the same issue :(

Things I have tried:

1. Power board via USB instead of 24v DC by adding jumper to VUSB
2. Try all 4 USB ports on Raspberry Pi 4B
3. Try 3x USB cables
4. Update klipper to latest (was about a month behind before) and recompile firmware
5. Update Raspberry Pi OS to latest
6. Run command 'dmesg' - nothing new found
7. Run command 'lsusb' - nothing new found

I am wondering if maybe the connection option is actually wrong and it should be something else? Or the CPU is actually a slightly different model?

It's clearly accepting a firmware so the board isn't dead but just not communicating via USB.

KrisWragg, 2023-03-13:
Looking at the schematic I think one thing I might try next (tomorrow?) will be to try and reuse the TFT connections rx/tx as UART.
https://github.com/bigtreetech/Manta-E3EZ/blob/master/Hardware/BIGTREETECH%20Manta%20E3EZ%20V1.0-SCH.pdf

Pins 3&4 for TFT can be used I think which would correspond to **Serial (on USART3 PD9/PD8))** in the make menuconfig settings.

Then can change klipper config to

```
[mcu]
serial: /dev/ttyAMA0
baud: 250000
restart_method: command

```

On Raspberry Pi I think you need to disable bluetooth by adding **dtoverlay=disable-bt** to /boot/config.txt

Then ensure that serial console is disabled:

```
sudo raspi-config
Select option 5 - Interfacing Options.
Select option P6 - Serial.
Select No - a serial console should not be available.
Select Yes - The serial hardware should be enabled.
```

KrisWragg, 2023-03-13:
So to clarify refer to:
https://github.com/bigtreetech/Manta-E3EZ/blob/master/Hardware/BIGTREETECH%20Manta%20E3EZ%20V1.0-PIN.png
https://cdn.sparkfun.com/assets/learn_tutorials/1/5/9/5/GPIO.png

TFT PD8 (TX) -> RPI GPIO 15 (RXD) aka Pin 10
TFT PD9 (RX) -> RPI GPIO 14 (TXD) aka Pin 8

KrisWragg, 2023-03-14:
Just a note, unless I am being dumb this isn't working either. Firstly it wouldn't compile klipper firmware with **Serial (on USART3 PD9/PD8))** selected.

I fixed this by editing klipper/src/stm32/stm32f0_serial.c and tweaked an existing if/def to properly define USART3_IRQn 

```
  #if CONFIG_MACH_STM32G0B1
    #define USART2_IRQn USART2_LPUART2_IRQn
    #define USART3_IRQn USART3_4_5_6_LPUART1_IRQn
  #endif
```

This compiled fine and installed on the board, but trying to use those pins is not getting me connected to it either. So I'm starting to wonder if this board is just duff or it's totally misdescribed either because it only functions with a CM4/CB1 or the settings are wildly different to those documented i.e. different chip???

bmorgan99, 2023-03-14:
If it was only ever intended for CB1/CM4 use, seems like they would want to clear that up as soon as possible.  That's a big oversight in my mind, and I looked hard for that before pulling the trigger.

Sonnenwerk, 2023-03-22:
I have the same issue..I got it as a replacement for my SKR Mini. But the board is not found on the usb port (using the same setup port/cable as before). Nothing i did (including steps above) helped !

KrisWragg, 2023-03-22:
I finally had a response to my ticket with them, it was as I suspected...

> The reason is this: On Manta series motherboards, the USB of the MCU is directly connected to the socket of the core board, and there is no connection to the type-c port. If you use a raspberry pie, it cannot be connected to the raspberry pie through a typeC cable and then communicate using USB, unless you use CB1 or CM4.
>Sorry for any inconvenience caused.
>BIQU

bmorgan99, 2023-03-22:
@KrisWragg 
Since it was a personal reply and neither the product page or this github was corrected, they must be happy with extra (unfortunate for us) sales.

KrisWragg, 2023-03-22:
Yes what they are saying is using a CB1/CM4 is mandatory not optional which is not as advertised. Will escalate my PayPal refund I think.

bmorgan99, 2023-03-22:
I will accept a CB1 as recompense :)

looxonline, 2023-05-07:
> Yes what they are saying is using a CB1/CM4 is mandatory not optional which is not as advertised. Will escalate my PayPal refund I think.

It is optional if you choose to run Marlin. This is why it is advertised as such.

bmorgan99, 2023-05-08:
It won't work with Octoprint if the USB port is unresponsive.  A very popular app.

bmorgan99, 2023-05-08:
Octoprint won't connect without a functional USB port.  Critical for many.

looxonline, 2023-05-08:
> Octoprint won't connect without a functional USB port. Critical for many.

Octoprint can connect via UART when using a USB to UART converter. The board is designed to host the SBC connection locally. If you want a board that hosts it externally then I would recommend the SKR Mini E3 or the SKR 3 EZ.

bmorgan99, 2023-05-08:
I would have had I known.  I had planned to stay with Marlin (printing noob), wait on CM4s to come down in price and go to Klipper.  That detail would have been nice to have.

looxonline, 2023-05-09:
> 

Why not just use the CB1 instead of wait for the CM4 to go down? The CB1 is a little more powerful than the Pi3B+ and more than enough to do all that klipper needs.

Jo0, 2023-05-10:
Product page mentions the use of a TPS5450-5A to help with powering external RPis. So in theory we should just be able to take our existing klipper setups, plug our Pi to the Manta, and update the printer.cfg to accommodate the new board. 

Plenty of us have existing setups and just want to upgrade boards for any number of reasons. (ie. Wanting to move away from my modified 4.2.7 board for dual independent Z, and utilize a board that has the feature built in.)

Sure, going to CB1 or CM4 directly is an option, but what if we want to save some costs upfront when the existing setup isn't broken or detrimental in any way.

looxonline, 2023-05-10:
> Product page mentions the use of a TPS5450-5A to help with powering external RPis.
> 
> So in theory we should just be able to take our existing klipper setups, plug our Pi (or thinclient like [#8 (comment)](https://github.com/bigtreetech/Manta-E3EZ/issues/8#issuecomment-1479550330) ) to the Manta, and update the printer.cfg to accommodate the new board.
> 
> Plenty of us have existing setups and just want to upgrade boards for any number of reasons. (ie. Wanting to move away from my modified 4.2.7 board for dual independent Z, and utilize a board that has the feature built in.)
> 
> Sure, going to CB1 or CM4 directly is an option, but what if we want to save some costs upfront when the existing setup isn't broken or detrimental in any way.

Yes, an external pi will work if you use it via SPI or UART which is not difficult to do and does not need any special adapters. You just wire the pi directly to the board using the UART lines. The product page is very clear, in many places, that this is designed primarily for use with the CB1 or CM4. The product description is the leader to the rest of the page and states this clearly.

[image]

If you still want to use it with an external pi then I would suggest going the UART route as it is very simple. Simply wire TX on the pi to RX on the board and vice versa. Ensure ground is hooked up and that you have configured the firmware to comm over the correct UART and at the correct baud and it will work without an issue.

rakosi2, 2024-03-09:
I have created a PR in Klipper that will allow the use of UART on the TFT pins.
In the `make menuconfig` select Communication interface to `Serial (on USART3 PD9/PD8)`
[image]

fleXXo22, 2024-12-25:
Has tehre been any news on this? I have been trying to follow everything done here, but i cant get the connection to the MCU. 
Please see the following Screenshots:
[image]
[image]
I also edited the /boot/firmware/config.txt and deactivated bluetooth.

rakosi2, 2024-12-26:
PD9/PD8 is on the TFT connector, please take a look at the pinout
[image]

fleXXo22, 2024-12-26:
As far as i can see this is connected correctly. 
I have it connected as follows:
Black/Gray Cable: Pin6 (GND) - TFT GND
Yellow/White Cable: Pin 8 (GPIO 14 TXD) - TFT PD9
Red/Purple Cable: Pin 10 (GPIO 15 RXD) - TFT PD8

Please see the attached Photos. Am I missing something here?
Thank you for your help.
[image]
[image]

rakosi2, 2024-12-26:
I used a FTDI cable cause I used a PC and not a pi. I would make sure that the baud rate is the same on both sides. For a RPi, you change it in cmdline.txt so change 115200 to 250000

fleXXo22, 2024-12-26:
My cmdline.txt says the following: `console=tty1 root=PARTUUID=a6f4af0e-02 rootfstype=ext4 fsck.repair=yes rootwait cfg80211.ieee80211_regdom=DE
`
Where do i have to add that?

rakosi2, 2024-12-26:
I would add:
console=ttyAMA0,250000
May have to reenable that uart service, not sure as I don't use a raspberry pi

fleXXo22, 2024-12-26:
I have added that and reenabled the uart service. Cant get a connection still. Do you have any other ideas I can try?

mikemerryguy, 2025-01-28:
> I have created a PR in Klipper that will allow the use of UART on the TFT pins. In the `make menuconfig` select Communication interface to `Serial (on USART3 PD9/PD8)` [image]

you are my HERO!!!
I just got two of these boards, and assumed the same as everyone else that I would be able to use a USB cable!!

I am running a repurposed Ubiquti Application server and thought the CM4 was nice, but really just liked the idea of this board and it was on a pretty good deal!

Then I get to the dreaded "cant mind MCU" no matter what, stumbled across this, ordered a couple FDTI cables off amazon, and BOOM reflashed with the usart3, and I now and controlling the beast from my liunx install!!!!

have yet to flash the other one, soon I will, have two printers getting these boards and have to print a plate to get it to sit where I want in the Sunlu S8 case and not short out the pins on the underside!!!

the other is a standard ender 3 so it will be easier but I only want to have one apart at a time, my bdsensors came today too so I am pretty excited,  The S8 might need some new steppers as they got wet and have some rust inside now haha

things been sitting for YEARS 

anyway, thanks again man!