bigtreetech/Manta-E3EZ #10: the guide doesnt show all the pins!!

MasterSe7en777, 2023-03-29:
im trying to add a dual Z stepper and i cant find the stepper pin so i can put it in the config, none of the stepper are labeled on the pinout please help

Valiance, 2023-04-19:
I am right behind you on this, I am trying to add my Z2, and am not finding any references that work for the Pinout.
Where did you find your pinout?
I know some of the pins would be identical for Z & Z2, I was planning to have independent steppers so I could program the dual Z auto-leveling feature.
I am stuck.   (Does not help I have never used Klipper before.)

DarKingF0x, 2023-04-20:
The dual z axis for this board is NOT independent. You should have been more clear on that. You can have two stuffer motors plug into those two spots but they're going to run at the same time at the same space they are not independent.
If you want dual independency you're going to have to convert your second extruder Port into a Z1 and label your 1st zport as Z0.
You will have to update your printer config.

MasterSe7en777, 2023-04-20:
Yeah sorry I didn't post my update, the z1 and z2 ports are tied together, if you look at the board traces they are both connected so basically only 1 driver controls them all I did was bump the current up to I believe 0.80 amps and it's totally fine driving both steppers

DarKingF0x, 2023-04-20:
What z motors are you using? Creality?

Tw0pay, 2023-09-08:
this is part of my config with Z1 taking place of E1 
`#   STEPERS Z  #

[stepper_z]
step_pin: PD2
dir_pin: !PD4
enable_pin: !PD3
microsteps: 16
rotation_distance: 8
endstop_pin: probe:z_virtual_endstop
#position_endstop: 0
position_max: 250
position_min:-15
homing_speed: 25
second_homing_speed: 1
homing_retract_dist: 2
homing_retract_speed: 10
homing_retract_speed: 5

[stepper_z1]
step_pin: PB7
dir_pin: !PB6
enable_pin: !PB4
microsteps: 16
rotation_distance: 8`

Valiance, 2023-09-09:
I resolved my Z & Z1 issues.   What I would up doing: I used stepper_z as default, then used E1 (Not E0) to set up Stepper_z1 for the second separate z rod control.  I used my pinouts for E1, and changed my printer.cfg file similar to what Tw0pay has listed above.  I also setup z_tilt, as per Klipper.3D's configuration https://www.klipper3d.org/Config_Reference.html Now, my Z rods will run separately, and tune themselves level.

daveneeley, 2024-08-17:
Just putting this here for the search engines...On Marlin for independent z-axis, change `Marlin/src/pins/stm32g0/pins_BTT_MANTA_E3_EZ_V1_0.h`

```
// Use the same Pins as Z2 (can only have one or the other configured)
// #define E1_STEP_PIN                         PB7
// #define E1_DIR_PIN                          PB6
// #define E1_ENABLE_PIN                       PB4
//#ifndef E1_CS_PIN
//  #define E1_CS_PIN                         PB5
//#endif

// Use the same Pins as E1 (can only have one or the other configured)
// There is a Z2_Motor connector on the board, but this is not for independent control
// It works the same way as the mini skr e3v3 (it just powers both motors from one driver)
// https://github.com/bigtreetech/Manta-E3EZ/issues/10
#define Z2_STEP_PIN                         PB7
#define Z2_DIR_PIN                          PB6
#define Z2_ENABLE_PIN                       PB4
#ifndef Z2_CS_PIN
  #define Z2_CS_PIN                         PB5
#endif
```