bigtreetech/Manta-E3EZ #15: Where is Marlin???

Tina-Builds, 2023-04-17:
You have this:
[image]
Where is the Marlin image???
How to I flash Marlin???

thisiskeithb, 2023-04-17:
The Manta E3 EZ (and entire Manta Series) was added to Marlin on February 7, so pull down the latest `bugfix-2.1.x` until `2.1.3` is tagged and released to use this board under Marlin.

You’ll define it as `BOARD_BTT_MANTA_E3_EZ_V1_0` in `Configuration.h`.

example:

```cpp

// Choose the name from boards.h that matches your setup
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_BTT_MANTA_E3_EZ_V1_0
#endif
```

Tina-Builds, 2023-04-18:
> The Manta E3 EZ (and entire Manta Series) was added to Marlin on February 7, so pull down the latest `bugfix-2.1.x` until `2.1.3` is tagged and released to use this board under Marlin.
> 
> You’ll define it as `BOARD_BTT_MANTA_E3_EZ_V1_0` in `Configuration.h`.
> 
> example:
> 
> ```c++
> // Choose the name from boards.h that matches your setup
> #ifndef MOTHERBOARD
>   #define MOTHERBOARD BOARD_BTT_MANTA_E3_EZ_V1_0
> #endif
> ```
> 
> > How to I flash Marlin, when Klipper is installed???
> 
> Nothing special is required. Simply copy `firmware.bin` to an SD card, insert it, and reboot the board to update just as you would under Klipper.

Oh no... I just need Marlin to see if the E3EZ then boots into DFU mode. in order to install CanBoot.. Just bad of BTT/Biqu to advertise "Marlin" runs  it and not to have an precompiled file here ...

docdawning, 2023-04-18:
> The Manta E3 EZ (and entire Manta Series) was added to Marlin on February 7, so pull down the latest `bugfix-2.1.x` until `2.1.3` is tagged and released to use this board under Marlin.
>
> You’ll [...]

@thisiskeithb thanks for the comment! So, just confirming if we place a Marlin firmware.bin built for this board on an SD card and boot with it, the board will automatically flash the Marlin firmware to the board? I'm planning on this board with Marlin without Klipper, at first. I have no CB1 or CM4.

Valiance, 2023-04-20:
Thank you for this thread.
I am seeking the Marlin info found for the Manta E3 EZ as well.   Glad to hear it has been added to Marlin on Feb 07.
I am struggling as to which files to get.

> The Manta E3 EZ (and entire Manta Series) was added to Marlin on February 7, so pull down the latest `bugfix-2.1.x` until `2.1.3` is tagged and released to use this board under Marlin.
>
> You’ll [...]

I did Download bugfix-2.1.x, both Marlin, and the Config folder.   I am not finding the board listed under the ...Config>Examples>Creality>Ender 3 folder anywhere.  I looked under several printers, but am not finding any reference to the board.  Here  is a pic of the Contents of the Ender 3 folder to explain better.

[image]

There are two E3 boards, but neither of them have the filename you mentioned: BOARD_BTT_MANTA_E3_EZ_V1_0

I know I am being smooth-Brained here, apologies.

thisiskeithb, 2023-04-21:
> There are two E3 boards, but neither of them have the filename you mentioned: BOARD_BTT_MANTA_E3_EZ_V1_0

`BOARD_BTT_MANTA_E3_EZ_V1_0` is the board name you'll use within Marlin.

There isn't a premade config for this board, so you'll have to build one out for now.

Valiance, 2023-04-22:
Thank you.   I did not know there was not a premade config.
I would have been barking up that tree for far too long.
I did not know one could even make their own premade configuration.
Now I guess it is a matter of locating instructions on how to do such a
thing.
Since there really is no Marlin version for this board, I am forced back
into Klipper again.
I am likely to not be the one who does the task of creating a Marlin config
for the board.
The main reason I am after a Marlin config is to gain true Pinouts for the
board.
The information given seems to be insufficient for each pin call out for
the whole board.

Regardless, thank you kind sir for the information.

thisiskeithb, 2023-04-22:
> I did not know one could even make their own premade configuration.

You can use the SKR Mini E3 V3 config as a starting point. Just update the `MOTHERBOARD` define and go from there. I have a test config, so I’ll try to get it cleaned up and submitted soon.

> The main reason I am after a Marlin config is to gain true Pinouts for the board. The information given seems to be insufficient for each pin call out for the whole board. Regardless, thank you kind sir for the information.

I built the pins file in Marlin from BTT’s pinout & schematics, but that’s also available for review: https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/Marlin/src/pins/stm32g0/pins_BTT_MANTA_E3_EZ_V1_0.h

Valiance, 2023-04-22:
Awesome, thanks!

thisiskeithb, 2023-04-24:
Here's the PR to add a Manta E3 EZ Ender-3 config to Marlin:
- https://github.com/MarlinFirmware/Configurations/pull/926

Kuzja27, 2023-05-17:
Has anyone installed Marlin on this board?  Technical support unequivocally answers me that this board is only for Klipper and, accordingly, it will not work without CB1 (or equivalent).  At the same time, this board appeared in the Marlin files.  I want to buy this board, but start working with Marlin and then move to Klipper.

Valiance, 2023-05-17:
No, I haven't.

Kuzja27, 2023-05-17:
Have you tried or installed Klipper directly? @Valiance

Valiance, 2023-05-17:
Yes, I did.
"Reality is the Ultimate Role-playing game." -- Victor A. Waters '88

Kuzja27, 2023-05-17:
@Valiance Too bad you couldn't install Marlin.  I really liked this board with the prospect of future upgrades to my printer.  I'll have to buy another board. I just don’t understand why add this board to the Marlin “Board.h” file if it is not installed there?

Valiance, 2023-05-17:
I agree with you.
I have had a ton of problems with this board

thisiskeithb, 2023-10-11:
> Has anyone installed Marlin on this board? Technical support unequivocally answers me that this board is only for Klipper and, accordingly, it will not work without CB1 (or equivalent). At the same [...]

@Kuzja27: Support must be working from old info. Manta boards will run Marlin without a CB1/CM4 with the only caveat being that you can't connect to the board externally via USB since the USB ports are connected to the CB1/CM4 and not the MCU.