bigtreetech/Manta-E3EZ #18: Klipper Screen on 3.5 Inch TFT Screen

Doormat1, 2023-04-29:
Hi, I have a 3.5 inch screen that plugs into the GPIO pins, I have installed the relevant drivers via terminal but whilst the screen lights up I get nothing displaying the screen. Is this feature possible on this board or do I need the BTT screen?

DarKingF0x, 2023-04-29:
Did you uncomment the tft settings in the boardenv.txt (root of sdcard)?

Doormat1, 2023-04-30:
No will give that a try

DarKingF0x, 2023-04-30:
I had the same issue where my screen was completely White, I shut down my system powered It down unplugged it took the SD card out put it back in my PC edited the BoardEnv.txt and removed the # infront of tftspi section. Saved and reinserted the chip and it works perfectly fine after that. It may take a few moments, to show up but it should work

AlexWild74, 2023-12-29:
Hello, I have also  a BTT E3EZ with CB1. I have lines on the display, but KlipperScreen does not start. Can anyone help

Doormat1, 2023-12-29:
What screen are you using? In the end I gave up with the gpio screen, (pin order on the board is not the same as a pi I believe was the issue) and used the btt spi TFT screen. Works fine now.

AlexWild74, 2023-12-29:
I used a wave share 3.5" HDMI  and at first I used the screen with the narrow flat cable (SPI). The result were the same. So I think the problem is not the screen.

Doormat1, 2023-12-30:
Have you removed the # from the line in boardevn.txt? Also check your ribbon cable, the connectors can be mirrored. I found that I needed a ribbon cable where the pads are on the opposite sides

AlexWild74, 2023-12-30:
Yes, I think I did that. Here is how my display looks. See attached pic.
[image]

Doormat1, 2023-12-30:
Ah it's on the log in page. Have you read this https://klipperscreen.readthedocs.io/en/latest/Troubleshooting/

AlexWild74, 2023-12-30:
sadly I could not find one of the listed issues in my log.