bigtreetech/Manta-E3EZ #20: Bootloader missing

gabber130779, 2023-05-10:
HI, bootloader has been erased. i cant flash with sd card .
Can you share bootloader.bin with guide for flashing to recover my board?

Tina-Builds, 2023-05-20:
BTT finally added it to the firmware folder :-)

closed 2023-05-27 by gabber130779

jbarron16, 2024-04-07:
Would we be able to get a guide on how to install the bootloader. I’m struggling on uploading the .bin file on the board. I have attempted WinSCP to the CB1 I have installed but run into permission issues for writing on the memory. Alternatively I’ve attempted manually adding the file to the BOOT SD card and running the line of code included but I receive genuinely errors and a failed flash .

PrintedVisionz3D, 2025-01-25:
I'm having the same issue, how to you transfer the bootloader bin to the CB1?

Tina-Builds, 2025-02-10:
> I'm having the same issue, how to you transfer the bootloader bin to the CB1?

The bootloader for the E3EZ has to be transferred to the E3EZ and not to the CB-1.
Just follow the instructions from: https://github.com/bigtreetech/Manta-E3EZ/tree/master/Firmware

5hiftyy, 2025-07-02:
I can't write the bootloader to the board either, even following those instructions on the page. It claims to have been completed successfully, but the bootloader and the firmware are both erased and neither are persistent after power cycling the board.