bigtreetech/Manta-E3EZ #28: Can't update MCU

Deapper, 2023-08-07:
I followed the instructions for flashing the mcu with little success;

First I compiled the firmware with the 'make' command after make menuconfig settings as sighted in the manual.
```
cd ~/klipper/
make menuconfig
```
Changed the klipper.bin to firmware.bin and loaded onto a sdcard root.

Placed sdcard in MCU slot
Power off/on even tried the reset swicth.
After checking the sdcard either time I can not see firmware.cur only firmware.bin remains

Deapper, 2023-08-08:
Is anyone else able to update MCU by sdcard?

Deapper, 2023-08-13:
Not sure why I can't flash by sdcard... Used DFU mode. To get into DFU mode press and hold button on Manta board while power on

closed 2023-08-13 by Deapper

5hiftyy, 2025-07-02:
I am running into this same issue, but even flashing via DFU won't stick after a power cycle.