bigtreetech/Manta-E3EZ #34: I upgraded to a Manta E3EZ and Melted the Y-axis Stepper Motor (housing) off my printer!

lugfug, 2023-12-09:
[image]
**The Problem:**
Let me tell you a charming story of melting plastic and destruction…

I have an “Ender 3 V2” with the OEM CREALITY control board and OEM Marlin. 

I was using it to reliably make prints a couple of times a week. 

The hardware upgrades are as follows: 
(All were installed and working with the OEM CREALITY control board and OEM Marlin.)
-  CREALITY Dual Z Motors (originally driven from one driver motor, via wire splitter, from the OEM control board.)
- Micro Swiss NG™ REVO Direct Drive Extruder
- BLTouch

**Need For Speed:**
I wanted to upgrade to Klipper to gain more speed, control over the printing process and Dual Z tramming.

So I installed a “BTT Manta E3EZ” control board with five “EZ2209” drivers and a “BTT CB1”

The physical install was painless and normal

I used the “printer-ender3.cfg” provided on the Manta E3EZ GitHub page and included it in my “printer.cfg”

So far So Good… or so I thought

**The "Great Melting":**
When I was configuring the BLTouch in the .cfg file, I started to smell the distinct aroma of melting ABS (not an electrical smell, nope, it was clearly melting ABS)… and the power supply fan unexpectedly turned on, even though the stepper motors were not moving. 

I thought _“hmm, that smell is odd… the power supply must be ‘burning off some dirt’…I wonder why the power supply fan is running…”_ **SMH** 🤦🏻‍♂️

I went back to configuring Klipper from my laptop. 

Then I realized the Y-axis motor would intermittently “quiver”, jiggling the print bed maybe a fraction of a millimeter. 

I thought _“hmm, that is odd, I wonder if my config file changes are causing it to spaz out… boy that ABS smell kinda stinks…”_ **SMH** 🤦🏻‍♂️ (not yet seeing the problem.)

I watched a few more minutes of my YouTube video for setting up the BLTouch and made a couple of `printer.cfg` edits. 

Trying to home the Y-axis this time revealed a “loose belt” and the Y-axis motor wasn’t moving the print bed. 

I tightened the Y-axis belt tensioner and no amount of tightening would make the belt tighten up. 

I thought _“hmm, that is odd, how did the belt get so loose?… Geez, that ABS smell REALLY stinks…”_ **SMH** 🤦🏻‍♂️ (STILL not seeing the problem.)

Looking under the bed I caught a glimpse of **_“movement”._** 

I thought _“Wait, what the heck is moving at the back of the printer…??!??!??”_

Just then, I see the Y-axis motor **FALL OFF THE PRINTER!**

**“Oh S#¡7 !!!, 

WHAT THA F*@#!!!!! 

GOD DAM*& IT!!!!!!, 

YANK THE F’ING POWER CORD!!!!!!!!!”**

There in front of me, laid the melted, contorted remains of my ABS, Y-axis motor mount… 
(pics below)

[image]

[image]

[image]

[image]

The stepper motor slumped to the table with a thud and the flaccid belt drooped helplessly over the carnage. 

**DOH!** 🤦🏻‍♂️

Still no electrical fire smell just melted ABS (which melts at 105 °C)

The motor was hot enough to give you a serious burn. 
 

Traumatized, I took the rest of the night off, no more 3D printers for me, _“GOD that ABS STINKS!”_

**Hardware Solution:**
Vowing to never repeat the “Great Melting of 2023”, I built a Y-axis motor mount out of a fence post that I bought at Home Depot which was 2”x2”, hollow, and MOST importantly it was ALL METAL!!! 
(pics below)

[image]

[image]

[image]

[image]

No more melting my motor off and yanking the power cord in a panic, LOL.

You can buy your very own 3 foot fence post to make your own ALL METAL motor mount here
 https://www.homedepot.com/p/2-in-x-2-in-x-3-ft-Black-Metal-Fence-Post-with-Flange-and-Post-Cap-58285247/324829989

**What Went Wrong???:**
Beats the F’ out of me… 

Maybe you know what I totally messed up. 

I’ll include my existing config file below for your review.

I’d love to make a few prints, or at the lease keep the damn thing from setting itself on fire!

**_Seriously though_**, would you look at my config file and offer your thoughts of how I could properly set up the printer … and avoid any further meltdowns?

THANK YOU for helping a fellow Maker in Need. 

**Lugfug**

NOTE: Here are the Model Numbers of the Stepper Motors just in case that is helpful. 

X:	BJ42D15-26V10 LOT NO.200525G 		(CREALITY part)
Y: 	BJ42D15-26V09 LOT NO.20006F		(CREALITY part)
Z: 	BJ42D15-26V10 LOT NO.200523D		(CREALITY part)
Z1:	BJ42D15-26V12 LOT NO.230306A1		(CREALITY part)
E: 	LDO-42STH25-1404AC(MSL) 221114	(LDO Motors part)

Current Config File:
```
# This file contains common pin mappings for the BIGTREETECH Manta E3EZ
# To use this config, the firmware should be compiled for the
# STM32G0B1 with a "8KiB bootloader" "8 MHz crystal"
# and "USB (on PA11/PA12)" or "CAN bus (on PB12/PB13)".

# See docs/Config_Reference.md for a description of parameters.

[stepper_x]
step_pin: PA14
dir_pin: !PA10
enable_pin: !PA13
microsteps: 16
rotation_distance: 40
endstop_pin: ^PC4
position_endstop: 0
position_max: 235
homing_speed: 50

[stepper_y]
step_pin: PC8
dir_pin: !PA15
enable_pin: !PC14
microsteps: 16
rotation_distance: 40
endstop_pin: ^PB0
position_endstop: 0
position_max: 235
homing_speed: 50

[stepper_z]
step_pin: PD2
dir_pin: !PD4
enable_pin: !PD3
microsteps: 16
rotation_distance: 8
endstop_pin: probe:z_virtual_endstop
#position_endstop: 0.0
position_max: 250
position_min: -6.0

[extruder]
step_pin: PD5
dir_pin: PD6
enable_pin: !PB3
microsteps: 16
rotation_distance: 33.500
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PB11 #HE0
sensor_type: ATC Semitec 104NT-4-R025H42G
sensor_pin: PA4 #TH0
control: pid
pid_Kp: 21.527
pid_Ki: 1.063
pid_Kd: 108.982
min_temp: 0
max_temp: 250

#[filament_switch_sensor material_0]
#switch_pin: PC5

#[extruder1]
#step_pin: PB7
#dir_pin: PB6
#enable_pin: !PB4
#heater_pin: PB10 # HE1
#sensor_pin: PA5 # T1

#[filament_switch_sensor material_1]
#switch_pin: PB1

[heater_bed]
heater_pin: PB2 #HB
sensor_type: EPCOS 100K B57560G104F #Generic 3950
sensor_pin: PA3 #TB
control: pid
# tuned for stock hardware with 50 degree Celsius target
pid_Kp: 54.027
pid_Ki: 0.770
pid_Kd: 948.182
min_temp: 0
max_temp: 130

[fan]
pin: PA8

[heater_fan fan1]
pin: PB15

#[heater_fan fan2]
#pin: PB14

[mcu]
serial: /dev/serial/by-id/usb-Klipper_stm32g0b1xx_22002D0002504B5735313920-if00

[temperature_sensor CB1]
sensor_type: temperature_host

[temperature_sensor MANTA_E3EZ]
sensor_type: temperature_mcu
min_temp: 0
max_temp: 100

[temperature_fan Case_fans]
pin: PB14
sensor_type: temperature_host
off_below: 0.4
min_temp: 10
max_temp: 90
target_temp: 55
control: pid
pid_Kp: 2
pid_Ki: 4
pid_Kd: 0.1

[printer]
kinematics: cartesian
max_velocity: 300
max_accel: 3000
max_z_velocity: 5
max_z_accel: 100

########################################
# TMC2209 configuration
########################################

#[tmc2209 stepper_x]
#uart_pin: PB8
##diag_pin: PC4
#run_current: 0.800
#stealthchop_threshold: 999999

#[tmc2209 stepper_y]
#uart_pin: PC9
##diag_pin: PB0
#run_current: 0.800
#stealthchop_threshold: 999999

#[tmc2209 stepper_z]
#uart_pin: PD0
##diag_pin: PC6
#run_current: 0.650
#stealthchop_threshold: 999999

#[tmc2209 extruder]
#uart_pin: PD1
#run_current: 0.800
#stealthchop_threshold: 999999

#[tmc2209 extruder1]
#uart_pin: PB5
#run_current: 0.800
#stealthchop_threshold: 999999

[board_pins]
aliases:
    # EXP1 header
    EXP1_1=PC1, EXP1_3=PC3, EXP1_5=PC0, EXP1_7=PA2, EXP1_9=<GND>,
    EXP1_2=PC2,  EXP1_4=<RST>, EXP1_6=PA0, EXP1_8=PA1, EXP1_10=<5V>

[bltouch]
sensor_pin: PA6
control_pin: PA7
x_offset: -30
y_offset: -10
samples: 2
speed: 2
z_offset: 0.0

[safe_z_home]
home_xy_position: 80,50 #Change coordinates to the center of your bed.
speed: 50 
z_hop: 10 #Move up 10mm
z_hop_speed: 5

#[output_pin PS_ON]
#pin: PA9

#[output_pin pb9_pin]
#pin: PB9

#[neopixel my_neopixel]
#pin: PC7

#[adxl345]
#cs_pin: PC15
#spi_software_miso_pin: PC11
#spi_software_mosi_pin: PC12
#spi_software_sclk_pin: PC10

[bed_screws]
screw1: 30.5, 37
screw2: 30.5, 207
screw3: 204.5, 207
screw4: 204.5, 37

[display]
lcd_type: st7920
cs_pin: EXP1_7
sclk_pin: EXP1_6
sid_pin: EXP1_8
encoder_pins: ^EXP1_5, ^EXP1_3
click_pin: ^!EXP1_2

[output_pin beeper]
pin: EXP1_1
```

dkustec, 2024-01-05:
Most humorous storytelling issue on GH I ever saw. 💕

billatgit, 2024-02-21:
remember to uncomment the lines (remove the #)
[tmc2209 stepper_y]
uart_pin: PC9
run_current: 650
stealthchop_threshold: 999999