bigtreetech/Manta-E3EZ #40: Bed wont heat

destruct123, 2024-05-24:
Just upgraded my ender 3 max. Everything works fine, but the bed wont heat even a celsius. think my board come dead.
[config-20240524-084117.zip](https://github.com/bigtreetech/Manta-E3EZ/files/15433692/config-20240524-084117.zip)

mast3rmindx, 2025-02-06:
Facing the same issue. Was it a dead board?

STATEDLIGHT, 2025-02-06:
Old post but just found it as I'm having same issue after around a year.  My bed isn't heating anymore.

mast3rmindx, 2025-02-06:
My first board had this issue fresh out the box, now the replacement board is having the same issue. Has to be a design flaw or bad quality checking.

STATEDLIGHT, 2025-02-06:
I've worked out why it's not heating, the fuse has blown.

mast3rmindx, 2025-02-06:
The 10AMP white mini fuse?

STATEDLIGHT, 2025-02-06:
15amp one next to the power for the bed

[image]

mast3rmindx, 2025-02-10:
I have replacement fuses I changed it still didn’t fix it for me

STATEDLIGHT, 2025-02-10:
My fuses haven't arrived yet but it's likely because I used too thin a gauge of wire between my PSU and the board.  I have my board in a case at the side of my printer.  About 3 or 4 days before the fuse blew I extended that cable so I suspect it wasn't thick enough and caused the fuse to blow so have ordered some thicker gauge wire also.  We will see 🤞

mast3rmindx, 2025-02-10:
Fingers crossed for you! I ended up using one of the fan ports to control an external mosfet for the heated bed so you have options if it doesn’t get fixed by replacing the fuse.