bigtreetech/Manta-E3EZ #42: Installing the Manta E3Ez board into the Ender 5 pro case.

Kuzja27, 2024-08-05:
Has anyone installed the board into the Ender 5Pro case using power supply adapters? I printed the adapter from the attached file in the Ender5 folder. If you have a couple of photos of the process or a description I would be very grateful.

samschultzponsys, 2024-09-25:
I am in the same boat, I do not really see how these stl brackets will help me with space in my E5

Kuzja27, 2024-10-03:
Here's a look at how I did it. You need to print two fasteners. The bolts (from the printer manufacturer) that are used to attach the power supply to the printer case, you use to attach the printed parts to the printer block. To attach the power supply with the printed parts to the printer case, you will additionally need 3 bolts and 3 nuts. Bolts M3 or M2 and approximately 15-20 mm long. Or use the bolts that you have available.
[image]
[image]
[image]
[image]