bigtreetech/Manta-E3EZ #43: Incomplete 3D Model of the E3EZ Board

BlueWings172, 2024-08-23:
Hello everyone,

I just downloaded the 3D model of the E3EZ board from this repository, but I noticed that it’s missing several key components, such as the Ethernet and USB ports and the stepper drivers. As I’m planning to design an enclosure for the board, having these details is crucial for accurate fitting.

Does anyone know where I can find a more complete 3D model that includes these components? Any help or pointers would be greatly appreciated!

Thank you in advance!

fischer-felix, 2024-10-20:
Same with me, I would also really like the complete model.

I have done some investigating with the pdf that is also provided. It is a pdf with embedded 3d model (viewable with Adobe Acrobat).

My efforts converting this pdf to anything useable have not really lead to much yet.

Ideally BigTreeTech would just share a complete step (or other common format) file of the product.

BTTJUSTIN, 2024-10-23:
Thanks for reporting. We will get the documentation updated and close this issue out once the updates are uploaded.

fischer-felix, 2024-10-31:
That would be great if you could upload the complete step file. 

Is it possible to get a rough timeline on this, as I am currently in the process of setting up my printer with an E3EZ, and it would be very helpful to have the 3d model to design a custom enclosure. 
So if it still may take some time, I might as well get started designing right away and make do with the incomplete model.

BTTJUSTIN, 2024-12-24:
Apologize for the late reply, Please check for link for the 3d model file
https://we.tl/t-D494Q5q2dA

BlueWings172, 2025-01-01:
@BTTJUSTIN 
Thanks for sharing the new model. I tried to downlead but the link doesn't work anymore. Would be possible to share the link again ? 

Many thanks

fischer-felix, 2025-01-01:
@BlueWings172 it seems the file in the GitHub repo in the folder 3D has also been updated, so you can just download it from there.

closed 2025-01-25 by BTTJUSTIN